lis-epfl/MAVRIC_Library #91: Position estimation with the Sonar

ndousse, 2015-05-21:
Dear @GregoireH  and @jlecoeur,

I implemented the Sonar in the complementary filter today. (in mixForExperiments branch, together with all pull request done lately). 

Here are some results and reflexion:
[image]
Actually, the vertical velocity given by the GPS is not so bad! In opposite, the barometer has quite some delay compared to the sonar. I already reduced the LPF gain of the velocity to reduce the delay. 
Proposition: reduce the gain of the barometer and increase the gain of the velocity. 

[image]
In position_estimation_correction, we interpolate the consecutive position from the GPS to estimate the velocity. Following this result, we see that it gives worst result than the velocity given directly by the GPS
Proposition: we use directly the velocity from the GPS in the position estimation

jlecoeur, 2015-05-21:
Nice! The altitude estimation looks quite stable now. Is it visible in flight?

Do you know how much computing the velocity from consecutive barometer and sonar readings is affecting the estimate?

And did you  give up with the kalman filter?

ndousse, 2015-05-22:
The first plot is actually obtained by computing the velocity from consecutive barometer and sonar readings, in my opinion, the sonar is improving the estimation as the barometer has some delay in the estimation and should have a lower gain correction. 

I don't want to invest too much time in the Kalman filter, maybe I'll get back to it later

jlecoeur, 2015-05-22:
I meant the effect of using barometer and sonar as velocity and position
sensors vs as position sensors only.
Because, by computing velocity from two consecutive position measurements,
you introduce noise that you integrate to get position again. I am
wondering if this not just making the estimate worse.

ndousse, 2015-05-26:
Here below some more results:
We can see on the first Fig. that the velocity estimation from the sonar is quite similar to the GPS velocity. There is not much delay, so the velocity estimation is better when using the sonar as sensor for the velocity
[image]

The barometer is slower when computing the velocity and more noisy. Therefore, we cannot reduced the LPF gain more... This estimation should have a much lower gain correction for the velocity. 
[image]

From what I observed, It seems that the "model" has to much weight, or that the correction is not sufficient. When there is correction from both the GPS and sonar the estimation is quite good, but as soon as one sensor is not giving an estimation (e.g. the sonar above 5m) the velocity estimation follows too much the acceleration integration and not the sensor measurements. 
[image]

This last fact can clearly be seen on this last Fig below where I had a (very) wrong biais estimation on the Z axis. The velocity estimation are quite consistent between themselves but the correction is not sufficient to pull the final velocity estimation due to a wrong integration of the acceleration. 
[image]

For the altitude, the results are quite nice. The GPS altitude diverges and we should continue not to take it into account in the altitude estimation (Kp_pos_gps[Z] = 0). 
[image]

Conclusion:
I would take a large gain for the velocity correction coming from the sonar and GPS, small for the barometer. (and have a correctly calibrated MAV).

closed 2015-07-15 by ndousse

itayguy, 2016-07-13:
Hi, 
where can I find ur complementary filter ?
and the paper with the equations ?

best

Itay

jlecoeur, 2016-07-13:
Hi !
You will find the complementary filter here : https://github.com/lis-epfl/MAVRIC_Library/blob/master/sensing/position_estimation.cpp
No equations sorry, but feel free to write them, we would add it to our documentation :-)

itayguy, 2016-07-15:
but it's not complementary filter it's more like alpha beta filter
and u will have some issue , with correction when the ground is at difrent them home position

jlecoeur, 2016-07-15:
Yes indeed this filter assumes the floor is flat.
During the landing procedure we give a constant downward velocity command
until the altitude stops decreasing, this compensates for varying floor
height.

itayguy, 2016-07-15:
u will get phantom velocity when u try to land on a different height . constant velocity may not help u , since u might get fault velocity.

jlecoeur, 2016-07-15:
what do you mean by phantom velocity ?

itayguy, 2016-07-16:
if ur estimated agl is 1 m
and then u find the ground at -10 m
what will happen for ur velocity estimator ?

jlecoeur, 2016-07-16:
We would have to check, I think not much would happen: we compute one velocity estimate per sensor, then we fuse them. The altitude is is not used to compute velocity, only the derivative of altitude measures. 
But I agree that the best would be to estimate absolute altitude AND ground altitude. I have a kalman filter in one of the dev branches that does this, I still have to merge it :-)