lis-epfl/MAVRIC_Library #96: Super weird uart communication with GPS

jlecoeur, 2015-06-18:
Hi guys,

I noticed very weird calls to `print_util_putnum` in gps_ublox.c
(see here https://github.com/lis-epfl/MAVRIC_Library/blob/master/hal/avr32/gps_ublox.c#L1302 )
This is strange because print_util_putnum tries to translate numbers in human readable format, does the GPS expect to receive a series of ASCII characters for each number?

Hopefully, I checked and all the values sent to the autopilot are not significatively converted by the function before being sent (ie all 16bits numbers where between 0 and 9!).
Even stranger, and this is where I start to wonder how it has ever worked, print_util_putnum prefixes positive numbers with a space (to align them with negative number on a console output)
( see https://github.com/lis-epfl/MAVRIC_Library/blob/master/util/print_util.c#L183 )
It means that the GPS receives all expected values, but separated by a ' ' character (decimal 32).
In any case, even if the GPS requires data to be sent in ASCII and separated by spaces, I think this is a bad idea to rely on the way numbers are formatted for console output.

I am really wondering if the GPS expects all this? I cannot test here because I do not have a GPS at hand. Can anyone try to compile the branch `dev_cpp` and plug a GPS in? I replaced the calls to put_num by direct calls to uart.
(see https://github.com/lis-epfl/MAVRIC_Library/blob/dev_cpp/drivers/gps_ublox.cpp#L1249 )

GregoireH, 2015-06-18:
I am still struggling to have it compiling on my windows machine :)

As soon as the branch compile I can test and report.

Best

ndousse, 2015-07-03:
Hi guys,
I just saw this conversation. The putnum function is used to send data from the autopilot to the GPS. 

It is indeed possible to configure the GPS via uart directly. 

To be honest, I believe it never worked and Julien's explanation of human readable data is probably the reason why. 

I never took any more time to test this and make it work because I didn't really need it...

jlecoeur, 2015-07-03:
All right, now it starts to make sense !
We may soon be able to configure the GPS, yeay  !

jlecoeur, 2015-10-07:
Solved by #114

closed 2015-10-07 by jlecoeur