lis-epfl/MAVRIC_Library #103: Remote control callbacks

basil-huber, 2015-08-10:
I propose an easy and clean way to register simple callbacks which are triggered as a remote control button is switched.
IT can be used as follows:
remote_callback_register(remote, my_cb_function, my_structure, remote_channel);
As the value of remote_channel changes, the function my_cb_function(my_structure, new_value); is called.

This allows to register callback without modifying the MAVRIC_Library. Furthermore, it is structured and clean. Furthermore, is "synchronized" with remote_update function, which avoids polling to often or not often enough

jlecoeur, 2015-08-10:
Sounds good !

Can we have a parameter to register the callback only for rising event or falling event or both ?

basil-huber, 2015-08-10:
Good idea to allow specification of rising/falling/both!

We could also add the remote_channel to the parameters given to the callback function, this would allow to have the same function on different buttons and to know which button has trigger the callback, but I don't no if it is really useful.

basil-huber, 2015-08-11:
I added rising/falling edge to the latest commit. However, at the moment the edge direction is not given to the function. Should it be given or should we leave that to the "user" ? (since the new value is transmitted, he could store the old value in the struct and compare)

There is an issue with the 11bit protocol: there is a tiny jitter on the button values (I have never seen the raw value to vary more than +/-1). If it really varies only between +/-1, than I will add that the change should be larger than 1. If it can be more, we could either define a common epsilon for all callbacks or let the "user" define an epsilon for each callback.

basil-huber, 2015-09-30:
Closing this issue in favour of issue #106.

closed 2015-09-30 by basil-huber