lis-epfl/MAVRIC_Library #105: Only channels 0-6 are available on MAVRIC

basil-huber, 2015-08-10:
It seems that only channels 0-6 of mavric are correctly connected to the remote control.
In both remote.channels and remote.sat.channels, only channels 0-6 can be controlled with the remote control (TURNIGY 9XR Pro). The remaining channels remain at 0.

Could this be due to a bug in spektrum_satellite.c:
https://github.com/lis-epfl/MAVRIC_Library/blob/master/hal/avr32/spektrum_satellite.c#L146

or does the problem lie at the remote control side?

basil-huber, 2015-08-10:
It seems that in spektrum_satellite.c:ISR(...) we only receive frame0 which contains channels 0-6

jlecoeur, 2015-08-10:
Double-check that your satellite and remote are bound in the proper mode:
https://github.com/lis-epfl/MAVRIC/wiki/First-Steps-Binding-your-Remote

GregoireH, 2015-08-11:
There were a note somewhere, in previous issue, that is now closed... That we should set the remote in proper mode before binding 
=> change default number of channels (8 I think) to something bigger IF NEEDED and set then as 11bits mode, since 10bits mode does not support it.
Can you report and add something in the wiki, if this solved your issue ?

Thanks

basil-huber, 2015-08-11:
It was really the channel limit on the remote. And everything is already nicely described in the wiki.

My bad!

closed 2015-08-11 by basil-huber