lis-epfl/MAVRIC_Library #106: Event / Callback system

jlecoeur, 2015-08-11:
#103 #104, made me realize we could use callbacks in other places (callback to new sensor value mainly).

There are limitations in the current implementations though:
- The callback function is called directly, and by-passes the scheduler. 
    * We have to be careful not to register long functions as callback functions, or tasks will be delayed. 
    * Even with short callback functions, we should not register too many for the same event, because they will be called at the same time.
    * There is even a risk of infinite loop if a callback function calls another callback function that ends-up calling the first callback function... no stabilisation = BOOM ;-)
- We need one implementation per type of event (mavlink message / mavlink command / remote event)

So why not adding an Event / Callback mechanism to the scheduler ?
A callback function would be a task. A task could be registered for a specific event.
When this event occurs, the next execution time of the task set to now. Then it is up to the scheduler to decide whether or not to run the task now, or to let the priority to another task.
We could even specify a special priority level for callback, for example higher than for normal periodical calls.

What do you think ?

basil-huber, 2015-08-11:
I like the idea of a central event handling.
I just think it should have a slim interface, rather than having an immensely generic thing, avoiding that it requires 100 lines and 100 parameters to print "hello world" as a button is switched,

jlecoeur, 2015-08-11:
I agree that right now, adding a task is heavy because we need to specify every single parameter (I guess this will be the same to create an event, and to register to task to en event). The C++ port should solve this.

For the interface, maybe we can make something similar to Qt signal/slot and event/callback (except we would schedule things):
http://doc.qt.io/qt-4.8/signalsandslots.html
http://doc.qt.io/qt-4.8/eventsandfilters.html

jlecoeur, 2016-06-07:
Following discussions we had, I close this issue because we preferred explicit dependency between modules.

closed 2016-06-07 by jlecoeur