lis-epfl/MAVRIC_Library #109: Disarming not possible if remote is trimmed

basil-huber, 2015-08-13:
Depending on the trim of the remote control, it can be impossible to arm/disarm the quad, since you might not be able to give the required input. Or, it might happen that disarming takes place unintentionally if trim is applied in the other direction.

We cannot get the remote's trim values on MAVRIC, right?

jlecoeur, 2015-08-13:
Nop, we only get the raw channel. Inside the remote, the signal is computed
as trim + stick position + whatever mix you programmed.

jlecoeur, 2015-10-07:
I think we can close this issue.

I recommend to let the trims centered. If you really need them and have issues to disarm, I suggest you create a mix on the remote so that when you flip a switch, it simulates the sticks in the corners and disarms.

closed 2015-10-07 by jlecoeur