lis-epfl/MAVRIC_Library #115: Build avr32-gcc from source

jlecoeur, 2015-09-17:
It is possible to build avr32-gcc from source (also for MAC OS)

I had issue with an update of glibc on my machine, the pre-compiled avr32-gcc was not working anymore.

I found the solution with this repository:
https://github.com/lis-epfl/MAVRIC/wiki/First-Steps-Setup-on-Linux

Happy to hear if someone can have it work on MAC OS :-)

dtsbourg, 2015-09-17:
Building from source works on MAC OS (tested on 10.10.4) thanks to the [avr32-toolchain](https://github.com/denravonska/avr32-toolchain). :+1:

ndousse, 2015-09-17:
I was also able to make it work on Mac os! :)

fschill, 2015-09-17:
> @lis-epfl/mavric-lis <https://github.com/orgs/lis-epfl/teams/mavric-lis>

awesome Julien, thanks for sharing! Finally I can use CodeLite natively
on the Mac :)

reminds me that I should release my mavlink-native bootloader some time
soon... works nicely now.

jlecoeur, 2015-09-23:
Indeed that would be great @fschill !

We often break USB connectors while flashing, a mavlink bootloader would be very handy!

GregoireH, 2015-09-29:
Hi Felix,

As discussed lat week, I can have a look and try to integrate within our
current MAVRIC code your stuff.
Thus, could you push mavlink-native bootloader somewhere please ?

It would be so nice to have it already :-) !!

Thanks !!

   Grégoire Heitz
   Support Engineer LIS Laboratory

fschill, 2015-10-07:
Hi, 

I was away the last few days. Anyway, the bootloader is now on github, with the boardsupport for the MAVRIC autopilot (we have several folders instead of just src/mavric, one for each hardware - the rest is identical). I copied the Library (just the necessary files), but usually this would be a submodule.
https://github.com/fschill/MavBoot

You should be able to compile it with CodeLite or AVRStudio (optimized profile to fit into 8kb). With UART it should be around 7500 bytes. 
There is also the XML file for the additional MAVLINK message definitions:
https://github.com/fschill/MavBoot/blob/master/dbgextensions.xml
The comments there should give a rough idea how it works. I still have to post the "other side" for sending images - it's essentially an extension to mavue, written in python. 
The bootloader will always activate at reset, send a mavlink message to say hello and wait for approximately 1 second for a reply. If no reply it launches the main program. The PC side, when active, will reply and switch the bootloader to boot mode where it will remain until the "start application" command is sent. 
The upload process consists of sending a number of binary blocks (32 bytes) until a full page has been transferred. Each message is acknowledged by the bootloader with a CRC of what it received. Once the full page has been transferred and acknowledged, the "write to flash" command is sent. The bootloader writes the page, and replies with an ACK containing the CRC of what is now in the flash. 
It is therefore all quite robust against dropouts, as the uploader can re-transmit blocks without ACK, and verify the pages directly after writing each page. I've successfully flashed through XBee, and also while forwarding from UART to an internal CAN-bus and back (it's a bit slower but works fine).

To speed up reflashing, multiple blocks can be sent in a burst while waiting for all the ACKs to come back. Also, I implemented an additional check before sending a page, where the uploader requests the CRC of the current flash page, and only sends the page if the CRC differs. This makes a huge difference if you only change a constant, as nothing will shift, and it's also a neat "Verify" feature (if the code is already uploaded, a re-upload is just a verify which takes 3-4 seconds). If code changes more significantly, the addresses will shift and the speedup is a lot less, but on average it's still a bit faster. If an upload fails, a re-upload will also resume where it failed.

More later... :) updated Mavue will follow soon...

Felix

fschill, 2015-10-12:
Hi again, 

just pushed my current version of Mavue to github: https://github.com/fschill/mavue

This has a Bootloader button that allows you to load a .hex file and flash it to the microcontroller via bootloader. 
First you have to compile and flash the bootloader to the mavric board. Start mavue with appropriate options, e.g.:
python mavue.py --device=/dev/ttyUSB0 --baudrate=115200
-click on "Bootloader"
- reset the mavric board - the Mavlink SYSID and COMP ID should now appear in the list of the new window
- select a flash file
- click "Write Flash"
It should now upload the file, and show a progress bar. Additional information (errors) are currently shown on the terminal output - check if there are errors. Ignore if there is a checksum error on the last flash page - this is normal as the page is not complete.
If you click "write flash" again it should be very quick (3 seconds) as it will only do a fast verify if the code is identical.

cheers, 

Felix

fschill, 2015-10-12:
..just noticed that this issue was closed - briefly re-opened to guide your attention to my last two posts with the bootloader and new version of mavue :)

ndousse, 2015-10-13:
Hi Felix,

Thanks for the two repositories! 
I am currently working on this and I would like to know whether there is a size limit on the binary we can send to the microcontroller? 
When I try to send it, I have mavue returning an error:
[image]

Do you have any clue?

ndousse, 2015-10-13:
In addition, on Linux, I have the following error if I choose any usb serial
```
Traceback (most recent call last):
  File "mavue.py", line 89, in update
    msg_key, msg=self.mavlinkReceiver.wait_message()
  File "/home/lis/Documents/myCopter/mavue/mavlink_receiver.py", line 141, in wait_message
    msg = self.master.recv_msg()
  File "/home/lis/Documents/myCopter/mavue/./pymavlink/mavutil.py", line 302, in recv_msg
    s = self.recv(n)
  File "/home/lis/Documents/myCopter/mavue/./pymavlink/mavutil.py", line 758, in recv
    ret = self.port.read(n)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 460, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected?)')
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected?)
```
did you ever have such issue as well?

fschill, 2015-10-13:
> python_error

Hi Nicolas,

I never had this error - seems to be happening within the intelhex
library that is included in the project. There isn't really a size limit
for the .hex files - the error rather seems to indicate a 32bit/64bit
issue or something like that (int size different).

Which OS does this error occur on? Is it 32bit or 64bit? I have mostly
tested it on 14.04 64bit without issues.

fschill, 2015-10-13:
This is also not an error I have seen. Sometimes if the power supply is
flaky the FTDI chip crashes - then something like this could happen. Or
maybe you have the serial port opened somewhere else?

ndousse, 2015-10-14:
Hi Felix,

The first error related to intelhex is occuring on Windows 7 64bit

The second error is occuring in ubuntu 14.04 64bit. 
I also tried mavue with the XBee and the standard LEQuad code and I got the same error. I received messages (e.g. heartbeat) for 3-5 seconds max and then the connexion is lost... 

I flashed the bootloader via the JTAG interface on Atmel Studio. I connected the board via the debug board and the 10 pin cable. I also tried to bootload via XBee but got the same issues. 

Do I have by chance to recompile pymavlink?
I use Python 2.7.6

fschill, 2015-10-14:
Ok - I never tested the bootloader part in windows I think. All the rest
(plotting, etc.) is working fine for me on all OSes. I did test again
the bootloader on Linux and MacOS - both worked fine for me without any
issues.

Regarding the serial connection - I think this must be related to either
the FTDI driver, or the python serial package. Do you have any issues
using the same hardware and Linux in e.g. qgroundcontrol? The only time
I have problems with serial is if the board is powered through USB and
draws too much power so that the USB port cuts off - this leaves the
FTDI driver in a bad state and I have to unplug and replug, and start
mavue again.
However if you have the same issue over XBee I assume the board is
powered separately. The XBee adapter is also FTDI, so maybe you have a
different driver than me?

pymavlink doesn't really have compiled parts, it's all pure python as
far as I know. The only exception is that it dynamically generates the
dialect code based on the XML message definitions if they don't already
exist - but the errors related to that if something goes wrong don't
have anything to do with the serial connection.

As I can't reproduce this error on any of my machines (2 Linux desktops
and MacOS) I don't really know how to proceed. Maybe try on a different
Linux machine to see if it's the OS or something else?

fschill, 2015-10-14:
FYI, on my current machine I have the following versions (from 14.04 repos):
python             2.7.5-5
python-serial   2.6.1build1
python-qt4       4.10 

To see the bootloader messages, you can also try to specify the dialect
with cmd-line option:
--dialect=dbgextensions
If the file is missing it should generate it - the bootloader needs it I
think. For our project we included the dbgextensions messages into our
normal message set.

jlecoeur, 2015-10-14:
Hi !

Thanks for sharing Felix.

Mavue works for me, but only with the baudrate at 57600:
```
python2 mavue.py --device /dev/ttyUSB0 --baudrate 57600 --dialect=common
```

Here are the versions on my machine:
**python2** 2.7.10 
**python2-pyserial** 2.7.5
**python2-pyqt4** 4.11.4-2

I haven't tried the bootloader.

fschill, 2015-10-14:
good to know, thanks for the feedback.

I've used baudrates 57600, 115200, 256000 and 512000 successfully - but 
it of course also depends on the settings and capabilities of the 
microcontroller. Maybe there's too much clock error at higher baudrates.
If not specified mavue uses 115200 as this is the default on all my 
projects - this can be changed in mavlink_receiver.py line 29.

When you compile the bootloader it makes sense to set the UART to the 
same baudrate you're going to be using in your project 
(mavric/boardsupport.h, currently it is at 115200). Otherwise there will 
be problems when switching between bootloader and application, as mavue 
currently doesn't change baudrate at runtime.

jlecoeur, 2015-11-04:
I close this because the compilation of avr32-gcc works, and let the discussion continue in #144

closed 2015-11-04 by jlecoeur