lis-epfl/MAVRIC_Library #122: Bad visualization on GitHub due to tabulations & spaces

ndousse, 2015-09-29:
On GitHub, the tabulations are completely messed up:
[image]
even if in the original file, it is perfectly aligned:
[image]

A potential solution would be to replace the tabs by a given number of spaces (e.g. 3). 
QGroundControl uses this formatting and the visualization on GitHub works fine. 

What do you think @lis-epfl/mavric-lis ?

basil-huber, 2015-09-29:
I think it is a good idea to use spaces! Most text editors have the option to automatically replace tabs with a given number of spaces while typing

jlecoeur, 2015-09-29:
OK for me as well. But please do not fix it before we decide what to do
with the cpp branch, that would be a disaster to merge back.

jlecoeur, 2016-02-02:
Anyone ready for a massive search and replace ? :-)

jlecoeur, 2016-02-03:
Will be fixed by #175

closed 2016-02-04 by jlecoeur