lis-epfl/MAVRIC_Library #126: Calls to functions of stdio.h in data_logging

jlecoeur, 2015-10-07:
Data_logging uses the function snprintf to concatenate strings. Functions of stdio.h are very big and we managed to avoid them most of the time.

When I try to remove the 4 snprintf calls, the space occupied by the program is reduced by 10% !

I propose to implement a short function to concatenate the file names in data_logging, and remove `#include <stdio.h>`

Any volunteer ? 
That is an "easy-fix", why not @kornat :+1: !

mdouglas90, 2015-10-20:
I'm writing a function to replace the snprintf and it requires parsing through the format specifiers (%[flags][width][.precision][length]specifier). There's quite a bit of options and only %s and %ld is used currently in the code. Would this be a function you plan to use a lot in the future or should I have the function only do the main ones?

jlecoeur, 2015-10-20:
If think two functions `data_logging_filename_append_int(char* filename, uint32_t i)` and `data_logging_filename_append_extension(char* filename)` would do the job here !

jlecoeur, 2015-10-20:
I don't know if we told you @mdouglas90 : we never push directly to master:
1. Create branch from master
2. Make changes, commit and push
3. Create pull request on the github interface
4. Assign someone and wait :-)

jlecoeur, 2015-10-22:
Fixed in #133

closed 2015-10-22 by jlecoeur