lis-epfl/MAVRIC_Library #140: Switch to COM_RC_IN_MODE onboard param for compatibility with new QGC

basil-huber, 2015-11-04:
The new QGC (v.2.7.1) requires the onboard parameter COM_RC_IN_MODE which indicates how the UAV is controlled:

-  0: RC transmitter
-  1: No control
-  2: Control over mavlink (i.e. Joystick)

c.f. https://pixhawk.org/firmware/parameters

basil-huber, 2015-11-04:
[image]

jlecoeur, 2015-11-04:
There is also a warning about `MISSION_COUNT` that is not answered by the UAV.
[image]

basil-huber, 2015-11-04:
For the COM_RC_IN_MODE onboard parameter:

Would changing the enum `source_mode_t` in `state.h` to
```
REMOTE = 0
GND_STATION = 1
JOYSTICK = 2
```

and changing the parameter name from `Remote_Src_Mode` to `COM_RC_IN_MODE` in `mavlink_telemetry.c`
do the trick?

jlecoeur, 2015-11-04:
Yes!

basil-huber, 2015-11-12:
COM_RC_IN_MODE is fixed in c7920d5cf1b88dd0ca449738c29fea66735d17d7 (already dev_cpp branch)

closed 2015-11-12 by basil-huber