lis-epfl/MAVRIC_Library #142: Minimum messages for compatibility with QGC

jlecoeur, 2015-11-04:
(Moved from #140)
Newer versions of QGC display a warning because the message `MISSIONS_COUNT` is not answered.

jlecoeur, 2015-11-04:
[image]

jlecoeur, 2015-11-04:
Mavlink provides a default implementation of the waypoint protocol
https://github.com/mavlink/mavlink/blob/master/missionlib/waypoints.c#L690
It seems our implementation is not complete, some message callbacks are missing. Anyone interested in fixing this @lis-epfl/mavric-lis ?

fschill, 2015-11-04:
Something I noticed some time ago: it seems the intention of the "waypoint list" seems to be to contain arbitrary commands (which can be waypoints). QGroundControl only supports a small number of them, but in principle it seems any command could be either sent directly for immediate execution, or appended to the waypoint list. Might be interesting to implement it this way, as it would add a lot of flexibility for mission planning and execution. Essentially, there would be a "command execution" module, which the "waypoint/cmd list" module feeds into (as well as direct commands).

jlecoeur, 2015-11-04:
Very good point!
Indeed [mavlink_mission_item_t](https://github.com/mavlink/c_library/blob/master/common/mavlink_msg_mission_item.h#L15) contains a GPS position and a MAV_CMD that can be anything. I think the intended scenario is a succession of "Go there / Do this" commands.
It means that the "Mission" module not only has to provide a list of cmds, but:
- Keep the list in sync with the ground station
- Give commands to the navigation (the "Go there")
- Monitor the position of the UAV (to check when the "Go there" is done)
- Notify the ground station
- Have someone execute the command associated with the mission item ("Do this"). This can be done via the command callbacks we already have.
- Check the result of the command (maybe notify the ground station)
- Loop to next mission item.

When receiving a new mission, the "Mission" module could decide to reject a mission item if the associated command cannot be handled (ie. no command callback was registered).

And for all the mission items that are not waypoints and do not require to move somewhere ("Do this" without the "Go there"), there is a [parameter defined in mavlink](https://github.com/mavlink/c_library/blob/master/common/common.h#L274)

fschill, 2015-11-04:
Exactly :)  Also, mavric-specific, there should be a clear separation between the navigation module and waypoint/mission module (I don't know how it is now, but at some point in the past there was a bit of convolution). My suggestion is:
- mission module sends the currently active waypoint or position hold point to navigation. It also checks if waypoints have been reached based on current position.
- navigation module navigates to one waypoint, and holds position when it reaches it. Output is a 3D velocity vector. 
- stabilisation takes the 3D velocity vector and performs the complete closed loop control

jlecoeur, 2015-11-04:
Yep it is still a nice mixture, and I agree with what you suggest.

jlecoeur, 2016-01-25:
Fixed in #145

closed 2016-01-25 by jlecoeur