lis-epfl/MAVRIC_Library #219: Send code version in AUTOPILOT_VERSION message

jlecoeur, 2016-04-19:
Would be very useful to check which version of the code is running on a robot. 

With #218, if the ground station sends a request with command REQUEST_AUTOPILOT_CAPABILITIES, the autopilot responds with a message of type AUTOPILOT_VERSION that currently contains zeros.

We could send 
- Code version (ex: v.1.5.2) with major=1, minor=5, revision=2.
- Git commit (ex: 0a3484ce) the first 8 characters of the git hash. This can be injected in code during compilation using a `-D GIT_HASH`.

ndousse, 2016-04-19:
Do you want to still incorporate this in PR #218?

jlecoeur, 2016-04-19:
No because the lack of callback for CMD  520 is really annoying when using dronekit, so I think it is better to merge #218 now to handle the command even if we do not answer with the code version yet.

I also think we still have to define how to fill the AUTOPILOT_VERSION message. So I would prefer another PR.

---
Here are the fields of AUTOPILOT_VERSION
```
capabilities                  uint64_t    bitmask of capabilities (see MAV_PROTOCOL_CAPABILITY enum)
flight_sw_version             uint32_t    Firmware version number
middleware_sw_version         uint32_t    Middleware version number
os_sw_version                 uint32_t    Operating system version number
board_version                 uint32_t    HW / board version (last 8 bytes should be silicon ID, if any)
flight_custom_version         uint8_t[8]  Custom version field, commonly the first 8 bytes of the git hash. This is not an unique identifier, but should allow to identify the commit using the main version number even for very large code bases.
middleware_custom_version     uint8_t[8]  Custom version field, commonly the first 8 bytes of the git hash. This is not an unique identifier, but should allow to identify the commit using the main version number even for very large code bases.
os_custom_version             uint8_t[8]  Custom version field, commonly the first 8 bytes of the git hash. This is not an unique identifier, but should allow to identify the commit using the main version number even for very large code bases.
vendor_id                     uint16_t    ID of the board vendor
product_id                    uint16_t    ID of the product
uid                           uint64_t    UID if provided by hardware
```

My suggestion:
- the uint32_t "version" fields for code version (ex 8bits for major, 8bits for minor and 8bits for revision)
- The uint8_t[8] "custom" fields for git hash
- flight_sw for the version of the project
- middleware_sw for the version of MAVRIC_Library

The issue I see is how to distinguish between projects, ex: is flight_sw_version the version of LEquad, or of MyCopter, Ywing, ... ?

ndousse, 2016-04-20:
We could also imagine having the submodules hash in the other custom_version flags.

jlecoeur, 2016-04-20:
Isn't the commit ID of the library enough to know which in which versions the submodules were ?

ndousse, 2016-04-20:
Good point! 

For the different project, I see two solutions: 
* A constant predefined value to differenciate the projects (eg LEQuad: 1, myCopter: 2, YWing: 1000)
* As we all probably have a repository for our projects, we can also give the hash of the commit for the project

jlecoeur, 2016-04-20:
> we can also give the hash of the commit for the project

Yes, I would put it in flight_custom_version

However it would be difficult to find which project it is only from the first 8 characters of the project commit ID.

Maybe we could use os_custom_version for the name of the project:
```
flight_sw_version             project version   ex: 0, 1, 1, 4 
middleware_sw_version         library version    ex: 0, 1, 5, 2
os_sw_version                 unused   
board_version                 board ID + version ex: 100, 4, 3, 1
flight_custom_version         project commit   ex: "40dcbdb0"
middleware_custom_version     library commit ex: "b6322b20"
os_custom_version             project name ex "LEQuad"  
```

PR #223 (2016-04-21): Add project name and git commit hash to message AUTOPILOT_VERSION
Fix #219
closed 2016-04-21 by ndousse