lis-epfl/MAVRIC_Library #224: moving altitude low pass filter from navigation to position estimation

GregoireH, 2016-04-25:
Hello,

Do you agree with me, that it is a bit weird to store the low pass filtered altitude estimation in navigation structure [cf here](https://github.com/lis-epfl/MAVRIC_Library/blob/master/control/navigation.hpp#L76) instead of in position estimation struct ?

if yes I can make a small PR for that.

basil-huber, 2016-04-25:
It is already in position_estimation_.local_position.pos[2].

I absolutely agree that it should not be stored internally in navigation, nor in waypoint_handler or anywhere outside of position estimation.

Would you mind waiting with the PR until I merged the classes?

GregoireH, 2016-04-25:
it is not the exact same as position_estimation_.local_position.pos[2], since it is a low pass fliter of that.

I will wait for that, 
let me know when I should proceed

ndousse, 2016-04-25:
This value is used during the landing procedure only. It used to be done in navigation, that's why it is still in the navigation structure. 

As it is now done in waypoint_handler, we can either move it to there or insert it in position_estimation

jlecoeur, 2016-04-25:
What is it used for ?

ndousse, 2016-04-25:
It is used at landing to have a smooth altitude variation to know when we touch the ground and we can switch off the motors.

jlecoeur, 2016-04-25:
Ok, then I think it is not part of position_estimation. The best would be in a (non-existing) take-off and landing controller.

Think of the position_estimation interface, it should be kept to the minimum and not expose many variables only because one module needs it.
We had the case with the attitude estimation. In the interface ahrs_t, the variables north_vector and up_vector were added because they are needed by stabilisation_copter. The problem was:
- north_vector and up_vector are redundant with the attitude quaternion
- every attitude filter needed to compute them
When @ndousse wrote a new attitude filter that estimated the attitude quaternion but did not compute north_vector and up_vector, it took time to find the issue.

basil-huber, 2016-04-25:
I suggest to meet all together and discuss a bit the whole structure before making small changes

jlecoeur, 2016-06-07:
I close this because
- this lpf is specific to the implementation of the landing procedure
- the interface of position_estimation is already large enough

closed 2016-06-07 by jlecoeur