lis-epfl/MAVRIC_Library #255: Dubin navigation

GregoireH, 2016-05-25:
Hello

If you are on a waypoint with auto next to false.
If you press stop, it will go back to the take-off position for some reason !?
Whereas it should remain on spot and change mode to "STOP_HERE", which requires a start to continue doing navigation.

@ndousse Could you have a look please ?

Thanks

ndousse, 2016-05-25:
I had a look and found a potential solution. 

What is exactly the expected behavior (for a wing its rather logical but not for a quad)? Should it stay on the spot, go towards the waypoint?

jlecoeur, 2016-05-25:
It should hover at its current location. For a wing, circle around the current location.

GregoireH, 2016-05-25:
exactly

ndousse, 2016-05-25:
For the quad, I agree.
For a wing, actually it depends. 
If you are already circling around a waypoint, you want to continue circling around this point as if you are currently on a line, you indeed want to circle around the current location.

jlecoeur, 2016-05-25:
I disagree. What you propose would be the good behavior if we ask to pause the navigation.

Edge case: the wing is circling a 500m waypoint, and you press STOP HERE because want to stop it before it collides into something. I would expect it to generate a new waypoint with the minimum radius, centered on its current location (or tangent to whatever). Certainly not continue its path around the waypoint.

GregoireH, 2016-05-25:
I do agree with @jlecoeur

ndousse, 2016-05-25:
Ok, solved in f4b26d1

ndousse, 2016-05-26:
Can we close this issue now?

jlecoeur, 2016-05-26:
Yes !

closed 2016-05-26 by jlecoeur