lis-epfl/MAVRIC_Library #259: Joystick commands not received

jacquemi, 2016-06-01:
I cannot receive Joystick commands: the callback that should receive the joystick values (Created in `joystick_telemetry_init`, in `joystick_telemetry.cpp`) has msgid 69 (`MAVLINK_MSG_ID_MANUAL_CONTROL`), but I'm only receiving messages with msgid 82 (`MAVLINK_MSG_ID_SET_ATTITUDE_TARGET`). Changing the value of msgid in `joystick_telemetry_init` did not solve the issue

ndousse, 2016-06-01:
Which version of QGroundControl are you using?

I believe in some of the version you can choose the type of messages sent by QGC for the joystick. Make sure that you choose the manual control type on the joystick toolbox.

jacquemi, 2016-06-01:
I'm using version 2.7.1, and yes, this was the issue, I changed it to manual and it worked. Thank you

closed 2016-06-01 by jacquemi