lis-epfl/MAVRIC_Library #265: Attitude estimation decays over time

mdouglas90, 2016-06-10:
The attitude estimation seems to decay to an incorrect value once it changes. For example, if you tilt the drone so that it has a roll of 90 degrees, the roll will change to around 90 degrees and then decay to around 40-50 degrees over a few seconds. If you set it to 45 degrees, it will decay to around 20-30 degrees. 

This does not appear to be the result of the weird magnetic field inside this building as it still occurs outside. In addition, the decay still occurs when the magnetometer update is commented out in ahrs_ekf.cpp.

jlecoeur, 2016-06-13:
@ndousse do you have a documentation for the ekf equations you used (or better how you derived them)?

mdouglas90, 2016-06-13:
In the mean time, I've written the code that sends the ahrs quaternion to the drone from the motion capture system. It currently works by writing over the ahrs quaternion every time the message is received. This also allows us to completely bypass the strange magnetic field that is in the dronedome. The code still calls the ekf update function so there is still some decay but it will jump back when it receives the next mocap message. If this decay is still not wanted and the update rate is not important, you can comment out the body of `Ahrs_ekf::update()` and the drone will just update the ahrs quaternion at 10 Hz. 

The branch locations for this code are:
- MAVRIC branch: `dronedome_ahrs`
- natnetclient branch; `dronedome_ahrs`

jlecoeur, 2016-06-13:
I opened #266 as the mocap does not really fix this issue.

ndousse, 2016-06-19:
Here is the two files (especially the first one) that I used to derive the equations for the EKF:
[Development_of_a_Planar_Low_Cost_Inertial_Measurement_Unit_for_UAVs_and_MAVs_(Fux2008).pdf](https://github.com/lis-epfl/MAVRIC_Library/files/322370/Development_of_a_Planar_Low_Cost_Inertial_Measurement_Unit_for_UAVs_and_MAVs_.Fux2008.pdf)

[Quaternions_3D.pdf](https://github.com/lis-epfl/MAVRIC_Library/files/322371/Quaternions_3D.pdf)

PR #269 (2016-06-21): Fix/ahrs ekf accel
Fixes #265. The issue seemed to be from the signs of part of the equations in the acceleration update. I rederived the equations and the drone was showing a much better attitude representation and wasn't decaying over time.
closed 2016-06-21 by jlecoeur