lis-epfl/MAVRIC_Library #275: Overshoot during navigation

mdouglas90, 2016-07-12:
There appears to be some less than ideal travel when using waypoints with at least the emulated drone. The picture below shows the flight path the drone took when making a 10 m x 10 m box. During the travel to the first waypoint, the drone overshot the waypoint by approximately 7 m before turning around to overshoot the waypoint again. I got the same results with both the NED and Global/Rel. Alt reference frames.

[image]

Has anyone else had any problems like this?

jlecoeur, 2016-07-12:
Never had this issue. Can you test in DIRECT_TO mode instead of DUBIN ?

mdouglas90, 2016-07-13:
It did the same thing in DIRECT_TO as well as DUBIN.

mdouglas90, 2016-07-13:
Here's something interesting I found that seems to be related somehow to the overshoot. The closer waypoint 0 is to the take-off location, the further the drone overshoots. I did some tests where I kept all of the waypoints the same except for waypoint 0, which I moved around.

[image]

[image]

[image]

These pictures show that as I move waypoint 0 forward, the drone overshoots waypoint 1 less.

---

I also tested this by moving the waypoint south and west as well. The pattern was the same, so it is not that waypoint 0 was closer to waypoint 1.

[image]

[image]

---

The picture that I showed in the original post contradicts this as there was overshoot moving to waypoint 0 instead (which is located 10 meters from the take-off location). However, that test had the altitude at 2 m instead of 10 m (The take-off altitude is set to 2 for all tests). So I tried a similar layout with global frame waypoints flying at 2 m and 10 m. 

[image]

[image]

The one flying at 2 m experienced overshoot whereas the one flying at 10 m did not. This implies that there is also an altitude effect on the issue as well

---

I tried another test where I put 2 consecutive waypoints on top of each other, but that did not cause any overshoot so it seems to be only near the starting waypoints:

[image]

closed 2016-10-14 by jlecoeur