lis-epfl/MAVRIC_Library #279: Non constant position command during landing

jlecoeur, 2016-07-18:
As noted by @GregoireH in #278 , during landing the XY position command is always reinitialised to the current position.

See https://github.com/lis-epfl/MAVRIC_Library/blob/master/communication/mavlink_waypoint_handler.cpp#L884

As a consequence, the drone may drift during landing.

GregoireH, 2016-07-19:
I would integrate that change in #278 otherwise we risk to have some useless conflicts ?

I can do that if you want.

There is not much to do according to me, except removing this line
https://github.com/lis-epfl/MAVRIC_Library/blob/master/communication/mavlink_waypoint_handler.cpp#L884
Since the set_auto_landing( ) already set the position to either our position or the specified one.

jlecoeur, 2016-07-19:
Ok! (just test it before you accept the PR)

closed 2016-10-14 by jlecoeur