lis-epfl/MAVRIC_Library #282: Command MAV_CMD_NAV_LAND not correctly implemented

jlecoeur, 2016-07-19:
Here is the documentation of mavlink command MAV_CMD_NAV_LAND used for auto landing:
```
21	MAV_CMD_NAV_LAND	Land at location
	Mission Param #1	Abort Alt
	Mission Param #2	Empty
	Mission Param #3	Empty
	Mission Param #4	Desired yaw angle
	Mission Param #5	Latitude
	Mission Param #6	Longitude
	Mission Param #7	Altitude
```

Our current implementation in Mavlink_waypoint_handler::set_auto_landing is not correct:
- it does not use param4 for the desired yaw angle
- it uses param1 to by-pass the param5 and param6: if param1=0 it lands on current location, if param1=1 it lands at coordinates given by param5 and param6.
- param7 is not used

GregoireH, 2016-07-19:
this was I think due to the fact that older version of QGC are not providing the 7 parameters
for eg. QGC v2.4.0 does not
So how you would handle that ?

jlecoeur, 2016-07-19:
We should focus on recent versions of QGC, which let you add a 'LAND' waypoint. I would first check the value of the params it sends.

GregoireH, 2016-07-21:
ok, @Spyroth Could you push your widget and the number of QGC version please ?
So that we could try to update the version we all use.

Spyroth, 2016-07-22:
The new widget has been pushed into Master branch. Its name is Control_Modes_Nav.qml
The QGC version I'm using is 2.9.5 (the newer versions show all the onboard parameters as integer)

GregoireH, 2016-07-22:
Thanks,

@lis-epfl/mavric-lis :
So I think we should now update the code, and people should use that version of QGC (2.9.5)
to remain compatible.

@Spyroth 
Do you mind taking care of adapting the code ?

Spyroth, 2016-07-25:
Done in #296

jlecoeur, 2016-07-25:
Not merged yet

closed 2016-10-14 by jlecoeur