lis-epfl/MAVRIC_Library #287: pb of fence while disabling imu calib in emulation

GregoireH, 2016-07-19:
cyril wanted to write some scripts to start the drone and run a small experiment several times to get statistics.

We found that we had to disable the imu start calibration,
https://github.com/lis-epfl/MAVRIC_Library/blob/master/sensing/imu.cpp#L68

But then the drone was getting into critical each time we were arming it.

I figured out that it was coming from the set_fence flag.
by [commenting out that line](https://github.com/lis-epfl/MAVRIC_Library/blob/master/sensing/position_estimation.cpp#L497), we can operate normally

I guess it is due to some problem between the timing where the origin is initialized

GregoireH, 2016-07-19:
@mdouglas90 Could you have a look please ?

mdouglas90, 2016-07-20:
There are 6 methods that can cause the drone to enter critical state:

- The battery is low
- The connection with the ground control station is lost
- The drone leaves fence 1
- The drone leaves fence 2
- The position estimation is unhealthy. Which means that either the barometers has not been initialized, the gps has not been initialized, or the gps is unhealthy.
- The connection has been lost with the remote

The issue appears to be related to the position estimation for the z direction. Without calibrating the imu, my z position estimation tended to increase by 300-400 m about ground (which surpasses the 75 m fence_1_z and the 100 m fence_2_z).

I don't know why the imu startup calibration would affect that as I don't believe that the simulated gyroscope uses any of the calibration values.

jlecoeur, 2016-07-20:
is the imu healthy ?

mdouglas90, 2016-07-20:
I can only find the function Imu::is_ready() which returns if the calibration is ongoing at the moment, not if the imu had a healthy read.

The issue is related to the barometer calibration which occurs during the first barometer read [here](https://github.com/lis-epfl/MAVRIC_Library/blob/ed1aa2993f78f6d71c4c54aaa5e008571a0a23f1/sensing/position_estimation.cpp#L162). For some reason that I don't know why, when I run it, the drone believes that it has an altitude of ~35 m belows the ground (so at 365 m above sea level). This function calls [this ](https://github.com/lis-epfl/MAVRIC_Library/blob/ed1aa2993f78f6d71c4c54aaa5e008571a0a23f1/drivers/barometer.cpp#L82)to set the altitude_bias_gf_. 

During the barometer update function, this variable is then subtracted from the measured value (altitude_filtered) in the [barometer_sim::update() function](https://github.com/lis-epfl/MAVRIC_Library/blob/ed1aa2993f78f6d71c4c54aaa5e008571a0a23f1/simulation/barometer_sim.cpp#L91).

When not calibrating the imu on startup, the first cycle of altitude filter is 0, so the bias is set to an addition 365 m off, which gives an altitude much higher than correct. After a cycle or so, the altitude filter would be correctly at 400 m. If the barometer were calibrated then, it would give a better altitude_bias_gf_. 

I assume that the issue does not appear when calibrating the IMU because the barometer already had several cycles to update its altitude_filter before we calibrate the barometer in position estimation. Position estimation will not call its position_correction function (which is where we first calibrate the barometer) until all calibrations are done. See [here](https://github.com/lis-epfl/MAVRIC_Library/blob/ed1aa2993f78f6d71c4c54aaa5e008571a0a23f1/sensing/position_estimation.cpp#L379).

I suppose a possible solution to this would be to ensure that the barometer has had at least one altitude_filtered before allowing calibration.

PR #289 (2016-07-22): pb of fence while disabling imu calib in emulation
This fixes #287 by creating two flags in the barometer class. One that is false until a measurement has been read and one that is false until it has been calibrated. A getter function returns the has_been_calibrated_ boolean so we can know if we should set position_estimation::barometer_init appropriately.

This has been tested in emulation.
closed 2016-07-22 by GregoireH