lis-epfl/MAVRIC_Library #301: Pb of fence + origin in real word experiments

GregoireH, 2016-07-26:
I was trying to do an experiment with Anand,

The drone was entering out of fence 2 custom mode, directly,
And its position on the map was centered on the config EPFL origin.

@jlecoeur and @mdouglas90 Could you have a look please ?

in emulation its working well.

We also had some pb in auto take off after disabling the fence 1 and 2 checks from the state_machine.cpp
The drone was not going up really well, but going up and side ways => crashes :(

However the wpt navigation is working well if you adapt the waypoint position to the EPFL origin.

Best

jlecoeur, 2016-07-27:
This seems related to #281. Fences are centered on origin (on EPFL esplanade by default), if they were centered on home, I think you would not have this issue.

> However the wpt navigation is working well if you adapt the waypoint position to the EPFL origin.

Can you be more precise ?

GregoireH, 2016-07-27:
The estimated drone position shown on QGC was centered on EPFL origin.
even though we were not there...

So in order to be able to do some wpt nav we had to set the wpt position relative to EPFL origin instead of on the correct place on the map.

Best

jlecoeur, 2016-07-27:
Found it. Position estimation has a member [origin_](https://github.com/lis-epfl/MAVRIC_Library/blob/master/sensing/position_estimation.hpp#L208), that shadows the [origin_](https://github.com/lis-epfl/MAVRIC_Library/blob/master/sensing/ins.hpp#L135-L136) member of its INS interface... 
So when Position_estimation updates the origin, it only updates its own origin, and waypoint_handler does not know about it.

GregoireH, 2016-07-29:
The origin Pb was solved in your former PR #303 

But the drone is still getting in Critical mode !
This because the fence position / configuration should be set somewhere, or all the the fence checks should be properly disactivate for now (which I personally disapprove)
=>

GregoireH, 2016-07-29:
I was testing with Anand and realize few Pb:
- The local position Z is not properly reset to 0 when you arm !

- The quad has a thrust != its min (when you arm with remote and then) after you switch to any of the velocity mode before actually taking off in manual
=> bad surprise for safety pilots and people around !

Maybe something is not initialized properly.
I checked that:
- the state.internal_state is set to NAV_ON_GND when we are in standby.

I would really advise to make in break in coding and spend more time testing,
before going any further ! @jlecoeur @basil-huber @mdouglas90 :)

Best

jlecoeur, 2016-07-29:
This a concern indeed, when you say bad surprise does it mean someone could have been injured ?

More testing is needed, for sure. But I disagree we should stop moving forward. On the contrary, I think efforts like #307 are promissing (at least in the description as I have not looked at the code yet) I think that will eventually reduce the frequency at which these stupid and dangerous bugs appear.

The problem is well summarized in #306, functionalities like navigation, fence, or critical are scattered around multiple files, with flags and states publicly accessible. These are modified at various occasions, each very important to avoid a crash.
This makes it very difficult not to introduce bugs for someone who has not written the code.
I would rather disable the problematic features and help @mdouglas improve the codebase. Then test it thoroughly outside.

closed 2016-10-14 by jlecoeur