lis-epfl/MAVRIC_Library #321: Cascade Controller layout

basil-huber, 2016-08-24:
@jlecoeur, @mdouglas90 and I decided to rewrite the controller structure and replace stabilization copter with a flexible cascade controller structure.

Last idea, we provide a templated 'cascade_controller' structure containing a controller for each level and 'mode_' member variable indicating the highest active controller level:

Cascade_controller<Nav, Pos, Vel, Att, rate>
{
public:
    enum  mode_t {POS_MODE, VEL_MODE, ... }
    void update()
    {
           switch(mode_)
           {
                  case POS_MODE:
                            // do position control
          }
    }
private:
    Nav_ctrl nav_ctrl_;
    Pos_ctrl pos_ctrl_;
    Vel_ctrl vel_ctrl_;
    Rate_ctrl rate_ctrl_;
    Att_ctrl att_ctrl_;

    mode_t mode_;
};

However, I find there are some limitations to this approach:

    The cascade has a fixed number of levels: using more or less levels would require changing the structure
    The modes are explicitly listed: adding modes requires modifiying the enum
    The 'cascade_controller' needs to explicitly expose all 'set_xxx_command()' functions and relay the command to the corresponding controller level

Keeping in mind safety and flexibility, I propose a new approach, where each controller inherits from is underlaying controller level, hence exposing its set function. The challenge of this approach is that the update function has to execute the correct levels. You can find my suggestion here:
https://gist.github.com/basil-huber/4f9f4ff579948129c729f5b5e2fc95c3

So, what do you think?

jlecoeur, 2016-09-27:
But I really like how you can combine controllers !

I think the problems are
- all controllers are fully template, this does not help readability and create big error messages
- they all have to call the correct update functions, this can be a source of errors.

Here is another version where the controllers are "normal", ie they can be used with or without cascade. 
They can be assembled together using a template class Cascade_inherit which takes two controllers as template parameters, and inherit from them.

https://gist.github.com/jlecoeur/c5095dd7b6f80645e039819c777ee9ec

What do you think @basil-huber ?

basil-huber, 2016-09-27:
First, this gave me even more of a headache than my solution. But after three aspirins, I start to love it!

jlecoeur, 2016-11-29:
solved with less templates in 471247f612fcdf3895c87b6979e65a06faff0475

closed 2016-11-29 by jlecoeur