lis-epfl/MAVRIC_Library #328: Integration of Mavric with Arduino library for GSM module

anandbaskaran, 2016-09-13:
I am in the phase of adding GSM support for the Mavric Library. For the communication with the cellular module, I have used a couple of files developed by Arduino (https://github.com/arduino/Arduino). Now I need to integrate these files with the Mavric, for which I need your suggestions.

Way 1: Just download the dependency that I have for operating and include it into the drivers.

Way 2: Make a submodule with Arduino library, so that we can use the Arduino  functions from MAVRIC. 

Way 3: Make a selective submodule only with the files that I require. (I think this is a good option. I am not sure on how to do it, but I will check this out if we are going with this option.)

jlecoeur, 2016-09-13:
I guess you are talking about this folder [drivers/2GCellular](https://github.com/lis-epfl/MAVRIC_Library/tree/dev/GSM/drivers/2GCellular)

It is strange that you write things like 
```cpp 
#include <Arduino.h>
```

I do not know your GSM module but I see two options:
- The GSM is an arduino board and this code has to run on the GSM module
    => create another repository

- You want the driver to be executed by the autopilot
    => remove all dependency to arduino

anandbaskaran, 2016-09-14:
You are right. We have to ideally go with option 2 of yours, as the module will directly connect with the mavric without using Arduino. As you said, we should remove the dependencies with Arduino, I will do that. 

However, I have some dependencies in the program like IPAddress.h which comes from Arduino library. What should I do in these cases? Can I directly use the (only necessary) headers into my 2GCellular (or) should I make a submodule, as these headers will be continuously updated.

jlecoeur, 2016-09-14:
It sounds overkill to have a submodule for a small class like arduino's IPAddress. Do you absolutely need arduino's implementation ?

anandbaskaran, 2016-09-14:
Yeah, I definitely need some headers implemented by Arduino. But as the existing thing works, it may not be very important to have a submodule for these files.

jlecoeur, 2016-09-14:
Ok, can you list the headers you need ?

anandbaskaran, 2016-09-15:
IPAddress.h -> Printable.h & WString.h ->PgmSpace.h (AVR Library)
Stream.h -> Print.h
SoftwareSerial.h
arduino.h (which I will Remove and merge with the libraries of mavric). This file has dependencies with io.h, Pgmspace.h and interrupt.h of AVR library, which Mavric should also have (i guess) .

jlecoeur, 2016-09-15:
Understand that we do not use Arduino and we do not run on the same architectures. The GSM driver will have to be adapted.
- **IPAddress.h** comes with IPAddress.cpp so you might have to reimplement what is inside the cpp file
- **Printable.h** I think you can easily get rid of this one
- **WString.h** have a look at our print_util.hpp and string_util.hpp to see if you cannot use it instead
- **SoftwareSerial & Stream.h** have a look at serial.hpp
- **io.h Pgmspace.h interrupt.h** will be useless

anandbaskaran, 2016-09-15:
Thank you! I will have a look into it.

closed 2016-11-29 by jlecoeur